CapibaraZero/fw #42: [FEATURE] ESP32-S3-BOX(-3) support

luciust, 2024-10-16:
There's a product called ESP32-S3-BOX and ESP32-S3-BOX-3 which is a ready developer kit - comes with touch display and exposed PINs. Would make a good base for the project. Don't want to link to the commerial pages, you can easily look it up
I

andreock, 2024-10-16:
Do you own one so you can test the firmware?

luciust, 2024-10-17:
Sure, I can even brick it, so let's have a go. The one I got is now considered obsolete (as the -3 replaced it) but it sports: ESP32-S3-WROOM-1-N16R8, 16 MB QSPI flash, 8 MB Octal PSRAM and a touch 2.4-inch display.
Let me check if I can flash it - if so we can later take a look at supporting that screen...
Example product page and datasheet: https://www.mouser.be/ProductDetail/Espressif-Systems/ESP32-S3-BOX

andreock, 2024-10-17:
[capibaraZero_0_5_0_esp32s3box.zip](https://github.com/user-attachments/files/17409447/capibaraZero_0_5_0_esp32s3box.zip)
Can you try to flash this firmware?

python3 esptool.py --chip esp32s3 --port UPLOAD_PORT --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size auto 0x0000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin

luciust, 2024-10-18:
OK:

`[  271.316927] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
[  282.255373] usb 1-2: USB disconnect, device number 6
[  287.802806] usb 1-2: new full-speed USB device number 7 using xhci_hcd
[  287.930574] usb 1-2: New USB device found, idVendor=303a, idProduct=1001, bcdDevice= 1.01
[  287.930580] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  287.930582] usb 1-2: Product: USB JTAG/serial debug unit
[  287.930584] usb 1-2: Manufacturer: Espressif
[  287.930585] usb 1-2: SerialNumber: 7C:DF:A1:A0:A9:A4
[  287.933797] cdc_acm 1-2:1.0: ttyACM0: USB ACM device
`
then:

`root@x260-dev:/home/dev/CapibaraZero/capibaraZero_0_5_0_esp32s3box# esptool.py --chip esp32s3 --port /dev/ttyACM0 --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin
esptool.py v4.7.0
Serial port /dev/ttyACM0
Connecting....
Chip is ESP32-S3 (QFN56) (revision v0.1)
Features: WiFi, BLE, Embedded PSRAM 8MB (AP_3v3)
Crystal is 40MHz
MAC: 7c:df:a1:a0:a9:a4
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 460800
Changed.
Configuring flash size...
Auto-detected Flash size: 16MB
Flash will be erased from 0x00000000 to 0x00003fff...
Flash will be erased from 0x00008000 to 0x00008fff...
Flash will be erased from 0x0000e000 to 0x0000ffff...
Flash will be erased from 0x00010000 to 0x00187fff...
Compressed 15104 bytes to 10429...
Wrote 15104 bytes (10429 compressed) at 0x00000000 in 0.3 seconds (effective 396.8 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 136...
Wrote 3072 bytes (136 compressed) at 0x00008000 in 0.0 seconds (effective 498.6 kbit/s)...
Hash of data verified.
Compressed 8192 bytes to 47...
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.1 seconds (effective 705.3 kbit/s)...
Hash of data verified.
Compressed 1539200 bytes to 931444...
Wrote 1539200 bytes (931444 compressed) at 0x00010000 in 17.3 seconds (effective 709.8 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...
`

luciust, 2024-10-18:
Connecting via picocom to the ttyACM0 at 9600 gives me loop of:

> Saved PC:0x4037763d
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x4bc
> load:0x403c9700,len:0xbd8
> load:0x403cc700,len:0x2a0c
> entry 0x403c98d0
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
> Saved PC:0x4037763d
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x4bc
> load:0x403c9700,len:0xbd8
> load:0x403cc700,len:0x2a0c
> entry 0x403c98d0
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
> Saved PC:0x4037763d
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x4bc
> load:0x403c9700,len:0xbd8
> load:0x403cc700,len:0x2a0c
> entry 0x403c98d0
> Killed

andreock, 2024-10-19:
[capibaraZero_0_5_0_esp32s3box_test2.zip](https://github.com/user-attachments/files/17445326/capibaraZero_0_5_0_esp32s3box_test2.zip)
Try this one please

luciust, 2024-10-19:
Different results:

Now the screen flahes white - black every second and I got on the console 'Amiga Stuff - Guru Meditation:

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0xc (RTC_SW_CPU_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
> Saved PC:0x421217f6
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x4bc
> load:0x403c9700,len:0xbd8
> load:0x403cc700,len:0x2a0c
> entry 0x403c98d0
> Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x42002580  PS      : 0x00060830  A0      : 0x82002656  A1      : 0x3fcebc60  
> A2      : 0x00000004  A3      : 0x3fcebc90  A4      : 0x00000000  A5      : 0x00000020  
> A6      : 0x00000010  A7      : 0x00000140  A8      : 0x80386b70  A9      : 0x3fcebcd0  
> A10     : 0x00000000  A11     : 0x3fcec1c4  A12     : 0x00000004  A13     : 0x3fcebc90  
> A14     : 0x00001800  A15     : 0x00000000  SAR     : 0x0000001a  EXCCAUSE: 0x0000001c  
> EXCVADDR: 0x00000008  LBEG    : 0x420c2af8  LEND    : 0x420c2b5c  LCOUNT  : 0x00000003  
> 
> 
> Backtrace: 0x4200257d:0x3fcebc60 0x42002653:0x3fcebc80 0x420024f9:0x3fcebd30 0x42002275:0x3fcebd50 0x42019ab2:0x3fcebd70
> 
> 
> 
> 
> ELF file SHA256: d95a6745b9f7a579
> 
> Rebooting...
>

luciust, 2024-10-19:
Hey, no problem, no need to aplogize, just push the test firmware, as mamy as needed. Flashing tomorrow!

luciust, 2024-10-20:
This time i got 45 degrees menu on the device, screen is working: 
[image]

andreock, 2024-10-20:
nice, next time I will add navigation button support and try to fix screen rotation. 

In future I will add also touch support maybe, yesterday I bought a touch display so it will be supported for sure

andreock, 2024-10-20:
[capibaraZero_0_5_0_esp32s3box_test4.zip](https://github.com/user-attachments/files/17450758/capibaraZero_0_5_0_esp32s3box_test4.zip)
Here you go, follow this pinout for the SD card:
- CS 10
- MOSI 11
- SCK 12
- MISO 13
I didn't put the navigation button for the moment.

luciust, 2024-10-20:
Thanks, will test - planning to solder SD card adapter next week...

luciust, 2024-10-20:
That gave me white screen no console... SD card not connected yet.

andreock, 2024-10-20:
I'm starting to think there's some problem with the serial. Try this one please. 
[Uploading capibaraZero_0_5_0_esp32s3box_test5.zip…]()

luciust, 2024-10-20:
Hope I'm not frustrating you - but test5 gave me black screen, no console activiity, just some reset info when tried to flash it again. Went back to test3 firmware just to check and it worked as previously with tilted 45 screen.

luciust, 2024-10-21:
Screen and orientation is OK with test6! SD card later...

luciust, 2024-10-21:
Connected the SD, nothing is blown, no activity on console tough!

andreock, 2024-10-21:
If you have SD card, can you test the capibaraZero_0_5_0_esp32s3box_test5.zip please? Test6 doesn't have SD initialization

luciust, 2024-10-21:
OK, going back to test5 gives me, regardless if SD card is powered or not powered via 3.3V

> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
> Saved PC:0x42024188
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x4bc
> load:0x403c9700,len:0xbd8
> load:0x403cc700,len:0x2a0c
> entry 0x403c98d0
> ESP-ROM:esp32s3-20210327
> Build:Mar 27 2021
> rst:0x8 (TG1WDT_SYS_RST),boot:0xa (SPI_FAST_FLASH_BOOT)
> Saved PC:0x42024188
> SPIWP:0xee
> mode:DIO, clock div:1
> load:0x3fce3808,len:0x4bc
> load:0x403c9700,len:0xbd8
> load:0x403cc700,len:0x2a0c
> entry 0x403c98d0
>

andreock, 2024-10-22:
Can you flash also test4, to see if serial give some debug information at least. thanks

luciust, 2024-10-27:
Hello,

tested both 4 and 5, no luck:

I had one partitnion on 4GB SD card - it hasn't been touched, also deleted it and no other changes happen after testing 4 or 5. Console is mute, but if I force file download - I get an ESP32 banner. So it is working presumably, just not telling me much.