CapibaraZero/fw #44: [FEATURE] Add Support for LILYGO T-Embed CC1101

tsbo, 2024-10-21:
Please add support for https://www.lilygo.cc/products/t-embed-cc1101
pin layout is shown at https://github.com/Xinyuan-LilyGO/T-Embed-CC1101/blob/master/examples/factory_test/utilities.h

(to my knowledge) the only difference is that it is using CC1101 instead of SX1276 for sub Ghz 

I have it, so I can contribute to testing FW. If needed, I can donate a hardware unit.

andreock, 2024-10-21:
Very interesting board, it has all the features that we need. I will buy one and start to work on it

tsbo, 2024-10-29:
i tried to apply update on new pins and startup sequence from factory test example of the project to "esp32-s3-devkitc-1" board, but this is still not enough and device is not starting up:
[my-changes.txt](https://github.com/user-attachments/files/17551736/my-changes.txt)

do you think something else is missing ?

andreock, 2024-10-29:
Do you have the stacktrace?

andreock, 2024-10-30:
Does the serial only give this error? Or is there also something like "guru meditation error"?

tsbo, 2024-10-30:
i am pretty sure i put here the guru meditation error, but it got lost somehow.

Update: i added sd card init, added a real card inside (tested it with factory image) and now the only error is 

[   606][E][sd_diskio.cpp:805] sdcard_mount(): f_mount failed: (

andreock, 2024-10-30:
Is display working?

-------- Messaggio originale --------

andreock, 2024-11-05:
I bought the board yesterday, as soon as it will arrive to me, I will start to work on it

YANGKUN87, 2024-11-16:
Great, we can add t-embed-cc1101 and wait for the firmware

andreock, 2024-11-16:
[image]
Here a little preview of the work, for the moment these feature works:
- WiFi
- BLE
- USB
- IR
- Network Attack
- Encoder
- Deep Sleep

YANGKUN87, 2024-11-17:
@andreock Hello sir, I have been paying attention to T-Embed CC1101 because I also have one. I can’t wait to try it now. Can you upload its BIN first? Thank you very much.

andreock, 2024-11-17:
Sure, I almost finished SubGHZ. When it will be ready I will publish the bin here. There is some minor bug with display in IR section and some crash but I will fix later

andreock, 2024-11-20:
Here you can download a preview firmware based on [this commit](https://github.com/CapibaraZero/fw/commit/64b8f46cc48c4af03f59affac332f319ef4bb34f)

[capibaraZero_0_5_1_lilygo_t_embed_cc1101.zip](https://github.com/user-attachments/files/17837803/capibaraZero_0_5_1_lilygo_t_embed_cc1101.zip)

You can flash using this command: `esptool.py --chip esp32s3 --port UPLOAD_PORT --baud 460800 --before default_reset --after hard_reset write_flash -z --flash_mode dio --flash_freq 80m --flash_size detect 0x0000 bootloader.bin 0x8000 partitions.bin 0xe000 boot_app0.bin 0x10000 firmware.bin`

Current issues:
- Encoder isn't really precise
- IR send have some glitches

YANGKUN87, 2024-11-22:
I just tried this firmware with t-embed-cc1101, and many features still need to be improved

andreock, 2024-11-30:
https://github.com/CapibaraZero/fw/releases/tag/0.5.1
Rotary encoder fixed, IR sender glitch fixed and various crash fixed.

closed 2024-11-30 by andreock