CapibaraZero/fw #49: [BUG] trouble with compilation

baptistereb, 2024-11-05:
**Describe the bug**
In platformio.ini, we call BLEScanner, which requires "ArduinoJson 6.21.4" to work, but the project needs ArduinoJson 7.2. When i try to "pio run", it fails.

**To Reproduce**
Just try to compile the project (but I'm not sure if I'm the only one experiencing this issue).

baptistereb, 2024-11-05:
it's fix, waiting for merge. See https://github.com/CapibaraZero/BLEScanner/pull/2

andreock, 2024-11-05:
Can you give me your compilation log? I just cloned fw repo from main branch and it compiles correctly

andreock, 2024-11-05:
fw repo compiles correctly but I'll merge since arduinoJson version changed in project

closed 2024-11-05 by andreock

baptistereb, 2024-11-05:
platformio run

Processing esp32-s3-devkitc-1 (platform: espressif32; framework: arduino; board: esp32-s3-devkitc-1)
----------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via -v, --verbose option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-s3-devkitc-1.html
PLATFORM: Espressif 32 (6.9.0) > Espressif ESP32-S3-DevKitC-1-N8 (8 MB QD, No PSRAM)
HARDWARE: ESP32S3 240MHz, 320KB RAM, 16MB Flash
DEBUG: Current (esp-builtin) On-board (esp-builtin) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES: 
 - framework-arduinoespressif32 @ 3.20017.0 (2.0.17) 
 - tool-esptoolpy @ 1.40501.0 (4.5.1) 
 - toolchain-riscv32-esp @ 8.4.0+2021r2-patch5 
 - toolchain-xtensa-esp32s3 @ 8.4.0+2021r2-patch5
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
rm -f *.cpp *.hpp *.o duckuino
bison -o parser.cpp -d parser.y
flex -o lexer.cpp lexer.l
Found 80 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 2.0.0
|-- Wire @ 2.0.0
|-- WiFi @ 2.0.0
|-- FS @ 2.0.0
|-- WifiSniffer @ 1.2.0+sha.4c0009b
|-- AppleJuice @ 1.0.0+sha.92e8bab
|-- SamsungBleSpam @ 1.1.0+sha.8876c46
|-- DHCPGlutton @ 1.1.0+sha.c1b0766
|-- SamsungBleSpam @ 1.1.0+sha.842d742
|-- HID @ 1.2.0+sha.ec42b93
|-- DuckyESP @ 1.1.0+sha.3f61a57
|-- NFCFramework @ 1.3.0+sha.23e0395
|-- PosixSD @ 1.1.0+sha.6995df7
|-- EvilPortal @ 1.12.0+sha.b37d377
|-- DHCPGlutton @ 1.1.0+sha.ca4680e
|-- BLESniffer @ 1.2.0+sha.ab22334
|-- BLEScanner @ 1.1.0+sha.268f73f
|-- ESPShark @ 1.0.0+sha.bb0dd50
|-- ARP_Poisoner @ 0.0.0+20241104221015.sha.3d41fd1
|-- IrFramework @ 1.0.0+sha.94fc35d
|-- GFXForms @ 1.1.0+sha.b5b2826
|-- ArduinoPcap @ 1.0.0+sha.d96bb01
|-- ArduinoJson @ 7.2.0
|-- AsyncTCP-esphome @ 2.1.4
|-- RadioLib @ 6.6.0+sha.9e06c4f
|-- battery_monitor
|-- UI
|-- Adafruit GFX Library @ 1.11.11
|-- Adafruit ST7735 and ST7789 Library @ 1.10.4
|-- LittleFS @ 2.0.0
Building in release mode
Archiving .pio/build/esp32-s3-devkitc-1/lib372/libwifi_attack.a
Indexing .pio/build/esp32-s3-devkitc-1/lib372/libwifi_attack.a
Compiling .pio/build/esp32-s3-devkitc-1/lib472/bluetooth_attacks/ble_attack_tasks.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/lib472/bluetooth_attacks/ble_attacks_btn.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/lib472/bluetooth_attacks/bluetooth_attacks.cpp.o
Compiling .pio/build/esp32-s3-devkitc-1/lib34e/sdcard_helper/sdcard_helper.cpp.o
In file included from lib/bluetooth_attacks/bluetooth_attacks.cpp:20:
lib/bluetooth_attacks/../../include/pcap_gen.h:2:25: warning: backslash-newline at end of file
 #define PCAP_FILE(type) \
                          
In file included from .pio/libdeps/esp32-s3-devkitc-1/BLEScanner/BLEScanner.hpp:21,
                 from lib/bluetooth_attacks/bluetooth_attacks.hpp:22,
                 from lib/bluetooth_attacks/ble_attack_tasks.cpp:20:
.pio/libdeps/esp32-s3-devkitc-1/BLEScanner/AdvertisedScannedDeviceCallback.hpp:24:7: error: deleted function 'virtual AdvertisedScannedDeviceCallback::~AdvertisedScannedDeviceCallback()' overriding non-deleted function
 class AdvertisedScannedDeviceCallback : public NimBLEAdvertisedDeviceCallbacks {
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .pio/libdeps/esp32-s3-devkitc-1/NimBLE-Arduino/src/NimBLEScan.h:20,
[... 91 lines cut ...]
 class AdvertisedScannedDeviceCallback : public NimBLEAdvertisedDeviceCallbacks {
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32-s3-devkitc-1/BLEScanner/AdvertisedScannedDeviceCallback.hpp:24:7: error: 'ArduinoJson::V6215PB2::JsonDocument::~JsonDocument()' is protected within this context
In file included from .pio/libdeps/esp32-s3-devkitc-1/ArduinoJson@6.21.5/src/ArduinoJson/Document/BasicJsonDocument.hpp:7,
                 from .pio/libdeps/esp32-s3-devkitc-1/ArduinoJson@6.21.5/src/ArduinoJson/Document/DynamicJsonDocument.hpp:7,
                 from .pio/libdeps/esp32-s3-devkitc-1/ArduinoJson@6.21.5/src/ArduinoJson.hpp:33,
                 from .pio/libdeps/esp32-s3-devkitc-1/ArduinoJson@6.21.5/src/ArduinoJson.h:9,
                 from .pio/libdeps/esp32-s3-devkitc-1/BLEScanner/AdvertisedScannedDeviceCallback.hpp:21,
                 from .pio/libdeps/esp32-s3-devkitc-1/BLEScanner/BLEScanner.hpp:21,
                 from lib/bluetooth_attacks/bluetooth_attacks.hpp:22,
                 from lib/bluetooth_attacks/ble_attacks_btn.cpp:20:
.pio/libdeps/esp32-s3-devkitc-1/ArduinoJson@6.21.5/src/ArduinoJson/Document/JsonDocument.hpp:286:3: note: declared protected here
   ~JsonDocument() {}
   ^
In file included from .pio/libdeps/esp32-s3-devkitc-1/BLEScanner/BLEScanner.hpp:21,
                 from lib/bluetooth_attacks/bluetooth_attacks.hpp:22,
                 from lib/bluetooth_attacks/ble_attacks_btn.cpp:20:
.pio/libdeps/esp32-s3-devkitc-1/BLEScanner/AdvertisedScannedDeviceCallback.hpp: In member function 'void AdvertisedScannedDeviceCallback::shrink_result()':
.pio/libdeps/esp32-s3-devkitc-1/BLEScanner/AdvertisedScannedDeviceCallback.hpp:28:24: error: 'class ArduinoJson::V6215PB2::JsonDocument' has no member named 'shrinkToFit'
     advertised_devices.shrinkToFit();
                        ^~~~~~~~~~~
Compiling .pio/build/esp32-s3-devkitc-1/lib683/ir_attacks/ir_attacks.cpp.o
*** [.pio/build/esp32-s3-devkitc-1/lib472/bluetooth_attacks/bluetooth_attacks.cpp.o] Error 1
*** [.pio/build/esp32-s3-devkitc-1/lib472/bluetooth_attacks/ble_attack_tasks.cpp.o] Error 1
*** [.pio/build/esp32-s3-devkitc-1/lib472/bluetooth_attacks/ble_attacks_btn.cpp.o] Error 1

baptistereb, 2024-11-05:
thank's for merging