CapibaraZero/fw #54: [BUG] Issue mounting sd card from esp32

XyDrive, 2024-11-30:
**Describe the bug**
When starting the Capibara, it failed to mount the sd card. It is well in fat32 and have the correct file structure from the script make_sdcard.sh 
The sd card reader is well wired from the good gpio include in the code.

**To Reproduce**
sd card reader GPIO :
SD_CARD_CS 13
SD_CARD_MOSI 35
SD_CARD_SCK 36
SD_CARD_MISO 37

**Expected behavior**
To have acces to the sdcard with the Capibara

**Stacktrace:**

log from screen when starting de Capibara: 

[  2334][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[  2606][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420124f4 arg=0x3fc9898c

Loop
Loop
[  8860][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  8865][E][sd_diskio.cpp:624] ff_sd_status(): Check status failed
[  9371][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Loop
Loop
#GUI inter in BadUsb
[  9876][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  9881][E][vfs_api.cpp:105] open(): /sd/ducky does not exist, no permits for creation
Loop

andreock, 2024-11-30:
Can you try if this firmware works for you? 
[capibaraZero_0_5_2_esp32-s3-devkitc-1.zip](https://github.com/user-attachments/files/17966876/capibaraZero_0_5_2_esp32-s3-devkitc-1.zip)

XyDrive, 2024-11-30:
Nop, not working sadly

#start screen and start capibara
[  2315][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[  2587][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420124f4 arg=0x3fc9898c
Loop
...
Loop #Open GUI USB rubber ducky
[  8099][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  8104][E][sd_diskio.cpp:624] ff_sd_status(): Check status failed
[  8610][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
Loop
[  9115][E][sd_diskio.cpp:126] sdSelectCard(): Select Failed
[  9120][E][vfs_api.cpp:105] open(): /sd/ducky does not exist, no permits for creation
Loop

andreock, 2024-12-02:
Can you send me your SD card socket model? Do you have ever tested it with sd card test sketch? I tested the firmware with my ESP32S3 and it works. The warning `spiAttachMISO(): HSPI Does not have default pins on ESP32S3!` refers to display since ST7789 display doesn't have MISO pin

XyDrive, 2024-12-02:
Ok i found the issue, i followed the eletrical diagram and not the pins.h 
The pins of the sd card reader was mixed with the ones of the sp12
[image]

actually there is a lot of wrong wiring in this diagram, i'm can redo it but not soon.

I'm happy it's working :D

andreock, 2024-12-02:
Diagram and definitions in pins.h seems to match for me.
In the diagram we have:
- SPI2_MOSI: GPIO 35
- SPI2_MISO: GPIO 37
- SPI2_SCK: GPIO 36
- CS: GPIO3

Definition of pins.h

`
#define SD_CARD_CS 13
#define SD_CARD_MOSI 35
#define SD_CARD_SCK 36
#define SD_CARD_MISO 37
`

XyDrive, 2024-12-02:
Yes but in the diagram the sd card pins "sck, MOSI, MISO" is mixed with the pins "sck, MOSI, MISO" of the SX1276

andreock, 2024-12-05:
Ok, I will fix it. Thank for reporting the error

andreock, 2024-12-05:
I separated SX1276 in a different SPI bus
https://github.com/CapibaraZero/resources/commit/61d9467d03b441f22ccd94aab7efb0b14c4970b8
https://github.com/CapibaraZero/fw/commit/48769390e4e4e2b1a28345e887360f93555fd57a

closed 2024-12-05 by andreock

sudosu0, 2025-01-16:
Hello, 

I'm trying build this for first time, and I'm running into the same error mentioned here. log:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce3808,len:0x4bc
load:0x403c9700,len:0xbd8
load:0x403cc700,len:0x2a0c
entry 0x403c98d0
[  2299][E][esp32-hal-spi.c:215] spiAttachMISO(): HSPI Does not have default pins on ESP32S3!
[  2571][E][esp32-hal-cpu.c:110] addApbChangeCallback(): duplicate func=0x420124f4 arg=0x3fc9898c
[  2881][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
[  2887][E][sd_diskio.cpp:806] sdcard_mount(): f_mount failed: (3) The physical drive cannot work
[  3198][E][sd_diskio.cpp:199] sdCommand(): Card Failed! cmd: 0x00
Loop
    Loop
        Loop
```