CapibaraZero/fw #55: Add layout Keyboard for Bad Usb

XyDrive, 2024-12-03:
The actual Bad USB is executing the scripts in the QWERTY US layout. It's not possible to execute ducky scripts for other machine that have other keyboard layout.

hints : https://github.com/hak5/bashbunny-payloads/tree/master/languages

andreock, 2024-12-05:
Actually official USBKeyboard library support some layout: https://github.com/espressif/arduino-esp32/tree/master/libraries/USB/src/keyboardLayout. I will add support in next release.

andreock, 2024-12-08:
Added support: 
https://github.com/CapibaraZero/HID/commit/cd1fac5b275201e395f0be829ffaac1cd532b985
https://github.com/CapibaraZero/USB
https://github.com/CapibaraZero/fw/commit/8fa93a01509dc804aa3423b42575061beb8bb3e6

Layout can be set by putting LAYOUT command at the beginning of a DuckyScript, for example

`
LAYOUT en_US
GUI r
ENTER
`

Supported layouts:

-  de_DE
-  en_US
-  es_ES
-  fr_FR
-  it_IT
-  pt_PT
-  sv_SE
-  da_DK
-  hu_HU
-  pt_BR

Further layouts will be added in future but will be tracked in another issues(an issue for every missing layout)

closed 2024-12-08 by andreock

XyDrive, 2024-12-10:
@Cardi8982 go to the discord server. I could try to help you from there

Cardi8982, 2024-12-11:
Thanks, I will.