CapibaraZero/fw #64: [FEATURE] #evil #portal save credentials

inv1sible, 2024-12-25:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is.

When playing (just copying the config and index.html examples) with the evil portal I realized that I was not able to retrieve the credentials

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

For starters I would like to see the creds in the serial monitor. It would be nice, if capibara saves the creds on the SD card and offers another HTML example to retrieve stored creds.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

As an example, please check solutions like bruce fw and or nemo fw for M5 devices

**Additional context**
Add any other context or screenshots about the feature request here.

Or am I am missing something here? In that case please provide guidance. Thanks and Merry Christmas

andreock, 2024-12-25:
Sure, I noticed that docs about EvilPortal is clearly insufficient and EvilPortal not working with Android 15. I will fix for 0.5.2. Thanks for reporting issue and Merry Christmas

andreock, 2024-12-25:
Do you have iOS or Android device? I tested my fixes on latest GrapheneOS Android 15 but I don't have an iOS device actually

andreock, 2024-12-25:
You can also new docs at: https://latest.docs-enu.pages.dev/docs/network_attacks/evilportal

inv1sible, 2024-12-26:
THANK YOU for yor swift reply and the updates. I did some further debugging, especially referring to your statement

> [...] EvilPortal not working with Android 15. I will fix for 0.5.2. 

I guess you mean that in Android 15 the phone is searching for index.htm*  in `generate_204` which has to be a directory within captive_portal like the output in serial monitor shows:

Android 15, Pixel6a
```
[611239][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204 does not exist, no permits for creation
[611256][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204.gz does not exist, no permits for creation
[611272][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm does not exist, no permits for creation
[611290][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm.gz does not exist, no permits for creation
[612094][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204 does not exist, no permits for creation
[612110][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204.gz does not exist, no permits for creation
[612127][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm does not exist, no permits for creation
[612144][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm.gz does not exist, no permits for creation
[612404][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204 does not exist, no permits for creation
[612420][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204.gz does not exist, no permits for creation
[612437][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm does not exist, no permits for creation
[612454][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm.gz does not exist, no permits for creation
[612475][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico does not exist, no permits for creation
[612491][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico.gz does not exist, no permits for creation
[612508][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico/index.htm does not exist, no permits for creation
[612525][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico/index.htm.gz does not exist, no permits for creation
[613607][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204 does not exist, no permits for creation
[613623][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204.gz does not exist, no permits for creation
[613640][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm does not exist, no permits for creation
[613658][E][vfs_api.cpp:105] open(): /sd/captive_portal/generate_204/index.htm.gz does not exist, no permits for creation```
```

Nevertheless, when manually opening http://192.168.4.1 in a browser you're getting there as in 

```
[195529][E][vfs_api.cpp:105] open(): /sd/captive_portal/index.htm does not exist, no permits for creation
[195546][E][vfs_api.cpp:105] open(): /sd/captive_portal/index.htm.gz does not exist, no permits for creation
```

_I know that the "does not exist, no permits for creation" is coming from vfs_api so and more over not showing an actual error but rather indicating, which files the webserver tried to open...._

**Unfortunately, there is no other output neither in the serial console nor in a creds.txt or whatever on the SD card :-( So could someone please check on this. THX**

inv1sible, 2024-12-26:
> Do you have iOS or Android device? I tested my fixes on latest GrapheneOS Android 15 but I don't have an iOS device actually

I am happy to help here. I did some captive portal tests with different devices / os versions

iOS18, iPhone12, not able to retrieve index.html

```
[ 48255][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html does not exist, no permits for creation
[ 48272][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html.gz does not exist, no permits for creation
[ 48289][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html/index.htm does not exist, no permits for creation
[ 48307][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html/index.htm.gz does not exist, no permits for creation
[ 48825][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html does not exist, no permits for creation
[ 48842][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html.gz does not exist, no permits for creation
[ 48860][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html/index.htm does not exist, no permits for creation
[ 48877][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html/index.htm.gz does not exist, no permits for creation
[ 48911][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html does not exist, no permits for creation
[ 48928][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html.gz does not exist, no permits for creation
[ 48945][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html/index.htm does not exist, no permits for creation
[ 48963][E][vfs_api.cpp:105] open(): /sd/captive_portal/hotspot-detect.html/index.htm.gz does not exist, no permits for creation

```

Android 11, Samsung S20FE, working here though

```
[825044][E][vfs_api.cpp:105] open(): /sd/captive_portal/index.htm does not exist, no permits for creation
[825060][E][vfs_api.cpp:105] open(): /sd/captive_portal/index.htm.gz does not exist, no permits for creation
[825337][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico does not exist, no permits for creation
[825354][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico.gz does not exist, no permits for creation
[825370][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico/index.htm does not exist, no permits for creation
[825387][E][vfs_api.cpp:105] open(): /sd/captive_portal/favicon.ico/index.htm.gz does not exist, no permits for creation
```

Isn't it that you decide, what the webserver returns... In case, the captive website isn't found just return the content from index.html or whatever you entered in config.json...... Or isn't that possible?

andreock, 2024-12-26:
https://github.com/CapibaraZero/fw/releases/tag/0.5.2-2024_12_26_55 You can test new firmware here, on android should be fixed

inv1sible, 2024-12-30:
Hi @andreock ... thanks for fixing that issue that fast!!! and sorry for the late reply. I forgot my device at my parents when visiting for XMAS. Anyway. I am happy to confirm it works: Captive Portal in A15 on a Google Pixel 6a showing and creds saved to SD card... #animageisworthmorethanathousandwords

[image]

closed 2024-12-30 by inv1sible