CapibaraZero/fw #67: [BUG] NFC ISO18092 scan locks up on lilygo C1101

fromport, 2024-12-26:
HARDWARE: lilygo with C1101
Software: 0.5.2 (beta)
**Describe the bug**
NFC menu, when I click on "Read ISO18092 tag" it switches to scanning mode.
Normally you can cancel and get back to NFC menu by clicking on the dial switch since "Go back" has a blue circle around it.
That doesn't seem to work, only way I can get control over it , is to press power button on the side of the lilygo (put it to sleep)
and wake it up again with the dial button on top.

**To Reproduce**
See above

**Expected behavior**
Just like with NFC "Read ISO14443A tag", when I click on the dial button, it should  return to NFC menu

andreock, 2024-12-28:
I switched back to good old capibaraZero's fork of Adafruit PN532 that seems better than Seeed-Studio PN532 library. For me the issue is fixed. Let me know if it's all fine on your side. 
[capibaraZero_0_5_2_lilygo_t_embed_cc1101_nfc_fix.zip](https://github.com/user-attachments/files/18264406/capibaraZero_0_5_2_lilygo_t_embed_cc1101_nfc_fix.zip)

andreock, 2024-12-28:
[capibaraZero_0_5_2_lilygo_t_embed_cc1101_nfc_fixes2.zip](https://github.com/user-attachments/files/18264442/capibaraZero_0_5_2_lilygo_t_embed_cc1101_nfc_fixes2.zip)
This one is final for the moment. It resets PN532 after polling to avoid PN532 becomes unresponsive under heavy usage

fromport, 2024-12-28:
fixes2.zip seems to have fixed this issue! I am now able to go back to the NFC menu again as expected.
Thanks for the quick response/fix!

closed 2024-12-28 by fromport