CapibaraZero/fw #68: [BUG] Wifi Scan doesn't seem to work on Liligo C1101

fromport, 2024-12-26:
Hardware: liligo CC1101
Software: 0.5.2 (beta)

**Describe the bug**
Main menu : I select Wi-Fi -> get 3 options: Scan/ Sniff / Go back
I press Scan
see scanning take place, % goes to 100%, see all the different channels increase, ends with channel 13
Display shows "No network found."
Screenshot WifiAnalyzer app on android
[image]

I have many IOT devices on 2.4Ghz, have 3 AP's with 2.4Ghz
Go back to menu, clock on "sniff"
Get on the screen
**Sniffing packets...
Sniffed packets: 1
[save]**

**To Reproduce**
See above

**Expected behavior**
See a list of available SSID's.
I saw it once. but never again.

stylesuxx, 2024-12-28:
Same hardware - scanning works fine on my end.
I can scroll through the results just fine - would prefer to see a list of results though instead of having "cards" of each of them...

Hardware: lilygo CC1101
Firmware: [0.5.2 (beta)](https://github.com/CapibaraZero/fw/tree/0.5.2-2024_12_28_52) slightly more up to date though.

fromport, 2024-12-28:
I just updated to the lated 0.5.2 beta (with the fix for #67 ).
Had to disconnect the battery so really from scratch, still nothing.
Could it be depending on the amount of AP's it find ? could you overload the results ?

andreock, 2024-12-28:
it seems to be a recurrent problem of ESP32. I don't find a real solution yet. I already do `WiFi.disconnect(); delay()` then start the scan but sometimes it stills fail(also to me).

andreock, 2024-12-28:
[capibaraZero_0_5_2_lilygo_t_embed_cc1101_def_wifi.zip](https://github.com/user-attachments/files/18267872/capibaraZero_0_5_2_lilygo_t_embed_cc1101_def_wifi.zip)
Can you try this one? It use recommanded settings by EspressIf(shorten channel time)

andreock, 2024-12-28:
[capibaraZero_0_5_2_lilygo_t_embed_cc1101_wifi_fix2.zip](https://github.com/user-attachments/files/18267943/capibaraZero_0_5_2_lilygo_t_embed_cc1101_wifi_fix2.zip)
This one seems better in my opinion. It implements a fix that I found on official Espressif Github and fixes a bug about GUI and WiFiScan that sometimes wasn't synchronized so that GUI printed "No networks Found" even if networks was found.

fromport, 2024-12-29:
While scanning it now "jumps" through all the channels and I can see them .
Nice.
I am going to try it a few more times and see if it will fo it all times.
So far I am really happy and impressed by your feedback and ability to fix things.
Thanks a lot !

closed 2024-12-29 by fromport