CapibaraZero/fw #70: [FEATURE] show running software version somewhere on the display of the lilygo CC1101

fromport, 2024-12-29:
**Is your feature request related to a problem? Please describe.**
I would like to be able to find out what firmware version I am currently running on my lilygo CC1101

**Describe the solution you'd like**
Some where in the menu ( below the battery % was suggested) make a menu with "about" or something that includes version info.

Not a deal breaker obviously, just better for support in the future as well.

fromport, 2024-12-29:
Feedback from Author on Discord

"taking screenshot of GUI display is really difficult so I don't think that I will add it shortly"

And that makes sense.

Don't know if I should close it, or keep it open and on the back burner.

andreock, 2024-12-29:
Keep it open, I will close it in future as not planned if necessary

andreock, 2025-01-03:
Can you check if latest pre-release is fine for the settings menu? Thanks

fromport, 2025-01-03:
Not able to compile the latest beta atm:
```
In file included from lib/nfc_attacks/flipper_zero_nfc_file_parser.cpp:22:
lib/nfc_attacks/nfc_attacks.hpp: In member function 'uint32_t NFCAttacks::get_version()':
lib/nfc_attacks/nfc_attacks.hpp:55:26: error: 'class NFCFramework' has no member named 'get_version'
     return nfc_framework.get_version();
                          ^~~~~~~~~~~
Compiling .pio/build/lilygo_t_embed_cc1101/lib1ff/network_attacks/network_attacks_btn.cpp.o
*** [.pio/build/lilygo_t_embed_cc1101/libfc7/nfc_attacks/flipper_zero_nfc_file_parser.cpp.o] Error 1
In file included from lib/nfc_attacks/nfc_attacks_btn.cpp:2:
lib/nfc_attacks/nfc_attacks.hpp: In member function 'uint32_t NFCAttacks::get_version()':
lib/nfc_attacks/nfc_attacks.hpp:55:26: error: 'class NFCFramework' has no member named 'get_version'
     return nfc_framework.get_version();
                          ^~~~~~~~~~~
In file included from lib/nfc_attacks/nfc_attacks.cpp:18:
lib/nfc_attacks/nfc_attacks.hpp: In member function 'uint32_t NFCAttacks::get_version()':
lib/nfc_attacks/nfc_attacks.hpp:55:26: error: 'class NFCFramework' has no member named 'get_version'
     return nfc_framework.get_version();
                          ^~~~~~~~~~~
*** [.pio/build/lilygo_t_embed_cc1101/libfc7/nfc_attacks/nfc_attacks_btn.cpp.o] Error 1
In file included from lib/nfc_attacks/flipper_zero_nfc_file_parser.hpp:20,
                 from lib/nfc_attacks/nfc_tasks.cpp:4:
lib/nfc_attacks/nfc_attacks.hpp: In member function 'uint32_t NFCAttacks::get_version()':
lib/nfc_attacks/nfc_attacks.hpp:55:26: error: 'class NFCFramework' has no member named 'get_version'
     return nfc_framework.get_version();
                          ^~~~~~~~~~~
*** [.pio/build/lilygo_t_embed_cc1101/libfc7/nfc_attacks/nfc_attacks.cpp.o] Error 1
*** [.pio/build/lilygo_t_embed_cc1101/libfc7/nfc_attacks/nfc_tasks.cpp.o] Error 1
```

andreock, 2025-01-03:
Run `pio pkg update` to upgrade NFCFramework

fromport, 2025-01-03:
[image]
[image]
**AWESOME**!!!

closed 2025-01-03 by fromport