CapibaraZero/fw #72: Create a AppsJS marketplace

andreock, 2025-01-18:
I think it can be a good idea to add a AppsJS marketplace, a website where anyone can share JS scripts/application made for capibaraZero. It MUST be:

- Serverless(use JSON or sqlite3 WASM as database)
- (Optionally) use Svelte or Vue as framework
- Divide each scripts in category that can also be filtered

A script must be accompanied with a description, a changelog and the link to source code. All this details will be fetched from the developer source code. This website won't save any details about scripts except link to a manifest(maybe in markdown or yaml format) that will contains all this informations and the name of the scripts.

xelab04, 2025-01-18:
Why specifically svelte/vue? I have experience with PHP + Laravel & Tailwind CSS + FilamentPHP and I think what you're trying to achieve would be really easy with that stack.

I can maybe make a mockup over the weekend if you want?

andreock, 2025-01-18:
The main problem of using PHP is that I would need a server to host it, using svelte or any JS framework(even vanilla JS) doesn't need a server to host the website and I can still use cloudflare pages.

blueandhack, 2025-01-21:
Another option is the GitHub Pages service. Vercel or Svelte are suitable for sharing JavaScript applications. I agree with Andreock; hosting a backend service could be more costly for an early-bird open-source project.

xelab04, 2025-01-22:
Alright, I get your point. On the flip side, if Capibara Zero ever needs servers, I have some free resources on mine 😄