CapibaraZero/fw #80: [FEATURE] SUB GHZ PROTOCOL

realdaveblanch, 2025-02-13:
**Is your feature request related to a problem? Please describe.**
I just bought this thing to experiment with subGhz, Bruce seems to fail a lot on the subghz zone, so im thinking about flashing capibara

**Describe the solution you'd like**
It seems like capibara only saves RAW signals instead of actually saving the protocol and the data.

**Describe alternatives you've considered**
No alternative, it seems like no project can actually save rf signals based on protocol (to later study it or bruteforce the data) but the FlipperZero

**Additional context**
Kind regards

andreock, 2025-02-14:
The actually implementation of capibaraZero the SubGHZ implementation, unfortunately, works even worst, i'm trying to fix it for next release.

realdaveblanch, 2025-02-19:
Sorry, just tested it and it was a ground loop problem, Bruce works great, I'll flash capibara and test it :)

Thanks for your time

closed 2025-02-20 by andreock